pingo-io/pingo-py #69: Support Intel Edison

Vido, 2015-02-06:
Thanks to Jomar (@homembit), Intel has kindly donated some Intel Edison samples.
We should support Intel Edison too.

rimolive, 2015-02-15:
I tried to run setup.py from the Github repo but I got this error:

    Traceback (most recent call last):
      File "setup.py", line 6, in <module>
        from setuptools import setup, find_packages
    ImportError: No module named setuptools

Searching in the opkg Edison repo, there is no python-setuptools package. If you guys know about other options to make Pingo run in Intel Edison I appreciate it.

homembit, 2015-02-15:
Use at your own risk: http://alextgalileo.altervista.org/edison-package-repo-configuration-instructions.html

-------- Mensagem Original --------

Enviado em:Sun, 15 Feb 2015 15:04:56 -0200
Para:garoa/pingo <pingo@noreply.github.com>
Cc:homembit <homembit@gmail.com>
Assunto:Re: [pingo] Support Intel Edison (#69)

ramalho, 2015-02-15:
On Sun, Feb 15, 2015 at 3:04 PM, Ricardo Martinelli de Oliveira
<notifications@github.com> wrote:
> I tried to run setup.py from the Github repo but I got this error:
>
> Traceback (most recent call last):
>   File "setup.py", line 6, in <module>
>     from setuptools import setup, find_packages [...]

Usually it's not hard to install setuptools directly (without using a
package manager). There is an ez_setup.py script that does the magic.
Take a look at instructions here:

https://pypi.python.org/pypi/setuptools/12.0.5

Best,

Luciano

-- 
Luciano Ramalho
Twitter: @ramalhoorg

Professor em: http://python.pro.br
Twitter: @pythonprobr

danilobellini, 2015-02-15:
You can install distribute/pip directly (probably without `--force-reinstall`): https://twitter.com/danilobellini/status/461653494295179264

rimolive, 2015-02-16:
Thanks @homembit but these repos are the ones I'm using. However, I decided follow @danilobellini suggestion and it worked.

Vido, 2015-09-14:
PR  #83 closes this issue

closed 2015-09-14 by Vido