pingo-io/pingo-py #72: pingo.detect.MyBoard() doesn't detect Arduino with Firmata with x86 Operating Systems

rimolive, 2015-02-16:
I'm running a x86 Windows system and Arduino can't be detected from this system. Need to find a way to solve this issue without affect the entire detection process.

ramalho, 2015-02-17:
On Mon, Feb 16, 2015 at 8:40 PM, Ricardo Martinelli de Oliveira
<notifications@github.com> wrote:
> I'm running a x86 Windows system and Arduino can't be detected from this system. Need to find a way to solve this issue without affect the entire detection process.

During the Campus Party I helped a person with this issue and it
turned out that Pingo *is* capable of detecting Arduino on Windows,
but for some reason the installation of PyFirmata using pip did not
bring the PySerial dependency, and the error reporting was wrong.

Try to manually install pyserial with Pip, then PyFirmata, and try the
detection again. Also, please report if you are trying to detect the
Arduino using detect.MyBoard or arduino.get_arduino.

Thanks!

-- 
Luciano Ramalho
Twitter: @ramalhoorg

Professor em: http://python.pro.br
Twitter: @pythonprobr

rimolive, 2015-02-18:
I'm using detect.MyBoard() and from I could see in the code the assumption is that the OS running must be 'x86_64' before detect Arduino. See https://github.com/garoa/pingo/blob/master/pingo/detect/detect.py#L50

Vido, 2015-02-18:
@ramalho , This issue is related with processor architecture, and not with operating system.
@rimolive , Thanks for finding the bug. I guess it's a easy bug to fix, can you make a pull request?

Vido, 2015-02-18:
Related #15