pingo-io/pingo-py #89: test_jumpwire fails on my Raspberry Pi 2 model B

lamenezes, 2015-09-13:
Changing the assert to LOW will fix this? This test assumes that some pins are plugged?
Details below

=================================================== FAILURES ==========================================================================
_____________________________________________________________________________ RaspberryBasics.test_jumpwire _____________________________________________________________________________
```python

self = <pingo.rpi.tests.test_rpi.RaspberryBasics testMethod=test_jumpwire>

    def test_jumpwire(self):
        ''' Wire this DigitalPin directly into VDD '''
        pin = self.board.pins[self.digital_input_pin_number]
        pin.mode = pingo.IN
        output = pin.state
```

    
\>       assert output == pingo.HIGH
E       AssertionError: assert 'LOW' == 'HIGH'
E         - LOW
E         + HIGH

pingo/test/level0/cases.py:64: AssertionError

ramalho, 2015-09-13:
@lamenezes thanks for looking into this. I can't reply in detail right now because I am swamped teaching full-time courses and reviewing the PT-BR edition of my book... However, in geral yes, there are tests in Pingo which require the boards to be connected to some simple circuits, otherwise we would have no way of telling whether a digital input is ever HIGH or LOW or changing... One of the biggest challenges ahead of us is designing and documenting these circuits, and building the actual test harnesses on actual boards for testing each release of Pingo. We have the boards, the components for the circuits are simple, we just have not properly designed, documented and implemented the harnesses in a reproducible manner. Please help us do that. Thanks!

lamenezes, 2015-09-14:
@ramalho Proper documentation would help on running and creating tests. And would also avoid misunderstandings like I had.
I was checking the tests and I saw that the button test was skipped because it required someone to be pushing the button while that exact test is running, how would we surpass that kind of problem? How would we automatically test an analog input such as a potentiometer?

I'm working on issue #68 and I should have a (simple) patch to submit soon. Therefore I don't have time to write docs right now.

Vido, 2015-09-14:
Indeed, testing is our Achilles heel.

@lamenezes , take a look in:
- pingo-py/pingo/test/level1/cases.py
- pingo-py/blob/master/pingo/test/level0/cases.py

We have some tricks to read a HIGH value and a analog input.
Sadly, they are described only in doc strings. 

Some Fritzing schematics would help a lot: 
https://github.com/pingo-io/pingo-py/tree/master/pingo/arduino/tests

hary66, 2015-09-17:
Hi everyone.

Sorry for polluting this post, but I've been asking a question a few days ago, and I'm not getting any answer.
So I wonder whether it's been seen by anyone ?
https://github.com/pingo-io/pingo-py/issues/91
I'm getting hard time trying using Pingo, and I'm seeking some help.
It would be great if somebody could say something......

Thanks.