pingo-io/pingo-py #91: New to pingo, can't get it working !

hary66, 2015-09-15:
Hi everyone.

As I said I'm new to pingo.
I' trying to use it with different card, but I can't get anyone working.

Currently, with the Arduino UNO I get this message 
Please wait while Arduino is being detected. This can take up to 30 seconds ...
Board Auto Discovery Failed!, Shutting Down
Traceback (most recent call last):
  File "essaiPingo", line 8, in <module>
    led = b.pins[13]
  File "/usr/lib64/python2.7/UserDict.py", line 22, in __getitem__
    return self.__class__.__missing__(self, key)
  File "/home/harold/pingo/pingo/util.py", line 61, in __missing__
    return self[self.normalize(key)]
  File "/usr/lib64/python2.7/UserDict.py", line 22, in __getitem__
    return self.__class__.__missing__(self, key)
  File "/home/harold/pingo/pingo/util.py", line 60, in __missing__
    raise KeyError(key)
KeyError: '13'

Any advice ?

Vido, 2015-09-17:
Hi @hary66 ,

Thanks for using pingo.

We have a mail-list, where the main contributors adresses this kind of problem:
https://groups.google.com/forum/#!forum/pingo-io
I recomend you to ask your question there. We consider Github's Issue tracker a bug/feature backlog.

Have you uploaded the StandardFirmata Sketch into to your Arduino?
Pingo can't talk with an Arduino without the Firmata protocol.

Sorry, I guess we haven't put enough emphasis on the documentation.

hary66, 2015-09-18:
Sorry to come back, but it seems I'm not authorized to do anything on
https://groups.google.com/forum/#!forum/pingo-io !

I can't open new question or answer any !  I guess I need  to be accepted by any kind of moderator ????

closed 2015-10-01 by Vido