pingo-io/pingo-py #93: Create a Debian package and submit to mentors.debian.net

turicas, 2015-09-28:
[pingo is not available on Debian yet](https://packages.debian.org/search?keywords=pingo&searchon=names&suite=all&section=all) and the first move to have it on official repositories is to create a package. Then, we need to find a Debian Developer to sponsor the packaging of this software and sign the package before uploading it to [Debian Mentors Server](http://mentors.debian.net/) so it can be approved (or not) by the FTP-Masters team and then enter in the official repositories.

@kretcheu, can you please do the packaging stuff? :) I can help with Python-related stuff.

kretcheu, 2015-09-28:
Hi, I began a Debian process of packaging, creating a ITP bug:
http://bugs.debian.org/cgi-bin/bugreport.cgi?bug=800393

kretcheu, 2015-09-28:
Hi, working on packaging to Debian, I saw that it's interesting thinks to do.
- changelog is recommended.
- signing tarball it's good practice.
- pingo/examples/dual7seg.py and pingo/examples/dualboard.py are identical.