pingo-io/pingo-py #96: Automagically create mocks (fakes) when the app is running outside a board

rodrigomaia17, 2015-10-19:
So, I think this feature is already partially implemented on the "ghost" directory. I would like to evolve this behaviour. Something like the gpiocrust lib works ( https://github.com/zourtney/gpiocrust/ ) . What is missing? How is this behaviour now? 

ps: Great lib! =D

Vido, 2015-10-28:
So, ghost is a fake interface to the 'no-existing-hardware'. In fact, ghost was create to enable our API testing. Comparing gpio_mock.py (from gpiocrust )and ghost.py (from pingo) they are similar mockups. What exactly do you think is missing?

For example:
When you want to test something on your computer, it's possible to run:
```
import pingo
board = pingo.ghost.GhostBoard()
```

rodrigomaia17, 2015-10-31:
Hey @Vido . Yes, but in this way I have to change the code when I want to run it in my computer. I think it can be more transparent. if I run `var board = ping.detect.get_board()` in my computer, will detect module instantiate ghostboard?  (if yes, then I misunderstood and it's already the way I'm suggesting )

ramalho, 2015-10-31:
On Fri, Oct 30, 2015 at 10:16 PM, Rodrigo Maia <notifications@github.com> wrote:
> Hey @Vido . Yes, but in this way I have to change the code when I want to run it in my computer. I think it can be more transparent. if I run var board = ping.detect.get_board() in my computer, will [...]

The behavior of get_board() currently is:

Plan A) detect whether pingo is running on a supported host;
Plan B) if not, detect whether there is an Arduino connected to a
serial port using the Firmata protocol;

Instantiating a ghostboard could be Plan C, however I am not 100% sure
that would be desirable. Failing with an error could be the best
solution, and a ghostboard should only be instantiated when explicitly
required.

Anyway, this is a minor issue. The ghostboard implementation needs a
lot of work that is more important than deciding what detect.get_board
does with it.

Best,

Luciano

-- 
Luciano Ramalho
|  Author of Fluent Python (O'Reilly, 2015)
|     http://shop.oreilly.com/product/0636920032519.do
|  Professor em: http://python.pro.br
|  Twitter: @ramalhoorg

rodrigomaia17, 2015-10-31:
Nice, thanks for the explanation @ramalho . Are there any issues for what is missing in the ghostboard? I would like to contribute (:

Vido, 2015-12-04:
@rodrigomaia17, I believe there is no issues related to ghostboard right now...

closed 2015-12-04 by Vido