pingo-io/pingo-py #100: Python 3 support

scls19fr, 2015-11-26:
Hello,

last source version of ` pingo-py` doesn't seems to support Python 3.

```
$ nosetests3
E
======================================================================
ERROR: Failure: ImportError (No module named 'board')
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/nose/failure.py", line 39, in runTest
    raise self.exc_val.with_traceback(self.tb)
  File "/usr/lib/python3/dist-packages/nose/loader.py", line 414, in loadTestsFromName
    addr.filename, addr.module)
  File "/usr/lib/python3/dist-packages/nose/importer.py", line 47, in importFromPath
    return self.importFromDir(dir_path, fqname)
  File "/usr/lib/python3/dist-packages/nose/importer.py", line 94, in importFromDir
    mod = load_module(part_fqname, fh, filename, desc)
  File "/usr/lib/python3.4/imp.py", line 245, in load_module
    return load_package(name, filename)
  File "/usr/lib/python3.4/imp.py", line 217, in load_package
    return methods.load()
  File "<frozen importlib._bootstrap>", line 1220, in load
  File "<frozen importlib._bootstrap>", line 1200, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1129, in _exec
  File "<frozen importlib._bootstrap>", line 1471, in exec_module
  File "<frozen importlib._bootstrap>", line 321, in _call_with_frames_removed
  File "/home/pi/github/pingo-py/pingo/__init__.py", line 2, in <module>
    from board import ANALOG  # noqa
ImportError: No module named 'board'

----------------------------------------------------------------------
Ran 1 test in 0.029s

FAILED (errors=1)
```

Is there any plan to support Python 3 ?

Kind regards

Vido, 2016-01-24:
Related #45

scls19fr, 2016-01-24:
It will be nice to first accept my PR about Enum. We will tackle Python 2 / 3 compatibility after merging Enum PR.