pingo-io/pingo-py #101: board = pingo.detect.get_board() or board = pingo.detect.MyBoard() ?

scls19fr, 2015-11-26:
Hello,

in http://www.pingo.io/docs/ we can see

    board = pingo.detect.MyBoard()

but in https://github.com/pingo-io/pingo-py/blob/master/README.rst we see

    board = pingo.detect.get_board()

When I see https://github.com/pingo-io/pingo-py/blob/master/pingo/detect/detect.py
I noticed

    def get_board():
        pass

    # TODO: deprecate legacy "MyBoard" factory name
    MyBoard = get_board

Thks to @lwalter86 for showing me this issue.

lwalter86, 2015-11-26:
good job 
thx

Vido, 2017-07-12:
Hi @scls19fr 

The correct function is  `get_board`.
The old `MyBoard` was kept for compatibility

@ramalho , Can you fix the pingo.io docs?

Thanks.